rwth-afu/UniPager #114: Support RasPi 3B+

dh3wr, 2018-04-27:
Zur Zeit "unknown RasPi detected". Brauchst du noch weitere Daten?

dh3wr, 2018-04-27:
[image]

7h0ma5, 2018-04-27:
Ich bräuchte die Revision aus `/proc/cpuinfo`, laut [elinux.org](https://elinux.org/RPi_HardwareHistory#Board_Revision_History) müsste das `a020d3` sein. Ich gehe mal davon aus, dass die GPIOs gleich geblieben sind?

dh3wr, 2018-04-27:
Auf der Platine ja, was die RasPi-Zuodnung angeht, so gehe ich auch von Gleichheit aus. cpuinfo kommt.

dh3wr, 2018-04-27:
pi@raspberrypi:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt                                                                                                                                vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4
processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt                                                                                                                                vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4
processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt                                                                                                                                vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4
processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt                                                                                                                                vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4
Hardware        : BCM2835
Revision        : a020d3
Serial          : 0000000072d2d5e9

7h0ma5, 2018-04-27:
Gefixt in https://github.com/rwth-afu/UniPager/commit/7b690281d419fb92647c0e5865c4a43703ef7a09. UniPager 1.0.1 müsste gleich im stable-Repository auftauchen.

closed 2018-04-28 by 7h0ma5