rwth-afu/UniPager #137: support for transmitter module CC1101

sp2ong, 2019-02-20:
It will be nice add support for transmitter module CC1101. It looks a nice solution for a personal small transmitter with WIFI network connected. We can take such a personal transmitter with you and for example in a hotel and any other place where there is a WIFI network, receive a message on your PAGER.
https://netxing.files.wordpress.com/2017/09/photo_2017-09-20_22-11-37.jpg?w=816 .

 Data sheet CC1101: http://www.ti.com/lit/ds/symlink/cc1101.pdf

The CC1101 module has 2-FSK, SPI communications.

dk7td, 2019-02-20:
Very good idea!
I discussed this with sp2ong already. Maybe these boards could be useful: https://www.ebay.de/sch/i.html?_odkw=cc1101&_osacat=0&_from=R40&_trksid=p2045573.m570.l1313.TR4.TRC2.A0.H0.Xcc1101+433.TRS0&_nkw=cc1101+433&_sacat=0

Will be much cheaper than a MMDVM if used for sending POCSAG only.

coolacid, 2019-05-01:
I ended up ordering 4 of these, so looking forward to this too

sp2ong, 2022-10-01:
Hi,

Good news that thanks to the author of the RadioLib library Jan Gromešb https://github.com/jgromes/RadioLib  support POCSAG protocol RX and TX 

[POCSAG](https://www.sigidwiki.com/wiki/POCSAG) using 2-FSK for modules:
SX127x, RFM9x, RF69, SX1231, CC1101, nRF24L01, RFM2x and Si443x

So we can build simple Receiver DAPNET based on SX127x, RFM9x, RF69, SX1231, CC1101, nRF24L01, RFM2x and Si443x devices