rwth-afu/UniPager #138: Calls with local:true cannot be decoded

dh3wr, 2019-02-21:
Local-Calls:
[image]

Normale Calls mit subscriber statt RIC:
[image]

7h0ma5, 2019-02-21:
Das kann ich leider nicht nachvollziehen, bei mir funktioniert das ohne Probleme:

```bash
curl -i -vvv --user dl2ic:**** -H "Content-Type: application/json" -H "Accept: application/json" -d '{"priority": 4, "data": "Hallo Welt", "local": true, "recipients": {"pocsag": [{"ric": 1234, "function": 2}]}, "distribution": {"transmitters": ["db0kwe"]}}' http://dapnetdc2.db0sda.ampr.org/calls
```

```
INFO - Message received: Message { id: "2c04eb9a-35bf-11e9-8e13-0242ac12000a", priority: 4, origin: "db0sda-dc2", expires_on: None, message: Pocsag(Message { mtype: AlphaNum, speed: 1200, ric: 1234, func: 3, data: "Hallo Welt" }) }
```

dh3wr, 2019-07-10:
Geht hier auch, closed.

closed 2019-07-10 by dh3wr