rwth-afu/UniPager #144: Rabbit-MQ-Version: Make compiled packages available

dh3wr, 2019-07-10:
Just for tracking

7h0ma5, 2019-07-10:
Dafür müssten wir die RabbitMQ-Version am besten in den master-Branch nehmen, dann wäre die testing-Version schon für DAPNET 2.0. Das Problem ist, dass bestimmt ein paar Leute das testing-Repository im Produktiveinsatz verwenden. Sonst müsste man wieder einiges an Jenkins-Konfiguration machen, damit es noch ein drittes Repository gibt für RabbitMQ-Version.

dh3wr, 2019-07-10:
Hmm. Das ist richtig. Ich könnte eine Email an alle Sender-Besitzer mit Unipager erzeugen, dass sie bitte das "testing" entfernen sollen.
Ist es viel Aufwand, ein 3. Repo einzurichten?

7h0ma5, 2019-07-10:
Das war relativ kompliziert, ich kann es aber mal versuchen.