rwth-afu/UniPager #145: rpitx als Transmitter-Device

do6uk, 2019-08-18:
Mit rpitx ist es möglich verschiedene Signale auf beliebigen Frequenzen direkt aus dem GPIO des Pi zu erzeugen.

Dieser Weg ist ohne zusätzliche Filterung der HF zwar nicht für einen Widerange-Sender nicht geeignet, aber für einen Hotspot oder Experimente im Shack dürfte es tauglich sein. 

Vom Prinzip her, könnte das Signal im Unipager, ähnlich wie beim Audio-Transmitter erzeugt und dann eben nicht auf das Sound-Device sondern an eine RPiTX-Pipe ggf. mit entsprechend notwendigen Umwandlungen weitergegeben werden.

Ich habe selber keine Ahnung von der Programmierung in Rust, aber soweit ich das erkenne, wird der Audio-Transmitter über aplay realisiert womit also grundsätzlich eine Anpassung an rpitx nicht unmöglich erscheint - ich habe nur keine Idee wie man das machen muss.

babooino, 2020-10-03:
At the very least, we should be able to use unipager as a client application to receive text messages from dapnet to file, or use some separate application for this purpose. Then we can use simple bash script to transmit this messages over rpitx. But in this case we will not be able to use dapnet multicast functionality to transmit POCSAG batches in specific timeslot. So we have to be able to run rpitx script exactly when assigned timeslot begins.

jbuchbinder, 2022-08-05:
You have the option of porting librpitx ( https://github.com/F5OEO/librpitx ) to Rust, or just compiling it as binaries and working that way. On limited hardware it might be more performant to port the code.

It seemed easier to just use bindgen to make bindings for the rpitx library, which I did :

https://github.com/jbuchbinder/librpitx-rust